thelastoutpostworkshop/esp32-2432S028_video_player #3: Scrambled video

JamieDull, 2025-07-02:
Hi, Thanks for your awesome tutorial. I followed along but when it plays back you video files, the video picture looks scrambled and is blocky? I did not use your exact Cheap Yellow board model, and my SD card is 512 GB.
I loaded all of the boards and libraries as shown. Any help would be appreciated.

Jamie

[image]
[image]

thelastoutpostworkshop, 2025-07-08:
The model seams to be ESP32-2432S022C.
The display control on this model is ST7789.
If you happen to have the pin information for the SPI and SD card reader, it would help.
I will research on my side if I can find the information.

thelastoutpostworkshop, 2025-07-08:
I found the schematic and the controller is ST7789 16 bit parallel:
https://www.amazon.ca/DIYmalls-ESP32-2432S022C-Touchscreen-ESP32-WROOM-32-Development/dp/B0DH1P13DW
The graphics library I use for this project do not have the 16 bit parallel support for ST7789, I suggest you open an issue, the maintainer is very active : 
https://github.com/moononournation/Arduino_GFX

JamieDull, 2025-07-08:
Thanks for your reply.  So if I were to buy a different Cheap Yellow Board, which processor should I look for?  Another user said he purchased from your Amazon link, and got the same issue I had.  Thanks for your help.

 

Jamie

 

https://www.amazon.ca/DIYmalls-ESP32-2432S022C-Touchscreen-ESP32-WROOM-32-Development/dp/B0DH1P13DW
https://github.com/moononournation/Arduino_GFX

thelastoutpostworkshop, 2025-07-08:
I bought mine from Alixpress : https://s.click.aliexpress.com/e/_oCVWMbC
I will order one from Amazon link I provided  to see what's going on (https://amzn.to/4epBJhD)

thelastoutpostworkshop, 2025-07-08:
I have the touch resistive model, probably some choose the capacitive model, more precise, but it comes with two different display controller, and the one with the ST7789 parallel controller is not supported by the library. I will add a note for this. 

[image]