thelastoutpostworkshop/esp32-2432S028_video_player #4: Erro na Instalação

AGNARCOLARES, 2025-07-04:
[image]

thelastoutpostworkshop, 2025-07-08:
Did you use the ESP32-S3 dev module ?

thelastoutpostworkshop, 2025-07-08:
Confirm you have this board type selected : 

[image]

The espressif boards version : 

[image]

And these libraries installed :

[image]

[image]

closed 2025-07-20 by thelastoutpostworkshop