thelastoutpostworkshop/esp32-2432S028_video_player #6: followed instructions, board does nothing.

Improbcat, 2025-07-23:
I appear to have a newer board than yours (mine only say ESP32-024), but was able to confirm the display has the correct "ILI9341 Display controller". When I follow your instructions it appears to do absolutely nothing. Screen doesn't light, video doesn't play. I have looked for pinout diagrams and have not found any I understand (I am very new at this). Certainly none with pin labels like your pin definitions in the code.

I think the issue might start at arduino IDE. I opened your sketch, and installed the required board and library managers of the correct version. then hit upload.

When compiling, the video shows many lines scrolling past in the output screen. On mine nothing appears there at all. When it finishes compiling and uploading uploads this is all that has appeared in the output:
---
Sketch uses 445610 bytes (33%) of program storage space. Maximum is 1310720 bytes.
Global variables use 41368 bytes (12%) of dynamic memory, leaving 286312 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.8.1
Serial port COM8
Connecting....
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 78:1c:3c:2c:5f:d0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x0007cfff...
Compressed 24976 bytes to 15972...
Writing at 0x00001000... (100 %)
Wrote 24976 bytes (15972 compressed) at 0x00001000 in 0.4 seconds (effective 483.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.0 seconds (effective 847.9 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 1008.9 kbit/s)...
Hash of data verified.
Compressed 445760 bytes to 242312...
Writing at 0x00010000... (6 %)
Writing at 0x0001beee... (13 %)
Writing at 0x0002e1ac... (20 %)
Writing at 0x000357e6... (26 %)
Writing at 0x0003af7a... (33 %)
Writing at 0x00040a2e... (40 %)
Writing at 0x00045d32... (46 %)
Writing at 0x0004b19c... (53 %)
Writing at 0x00050831... (60 %)
Writing at 0x00055a3b... (66 %)
Writing at 0x0005bf28... (73 %)
Writing at 0x000663d0... (80 %)
Writing at 0x0006d441... (86 %)
Writing at 0x000729fe... (93 %)
Writing at 0x00078196... (100 %)
Wrote 445760 bytes (242312 compressed) at 0x00010000 in 4.6 seconds (effective 773.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
---

Help? I have no idea where to start.

Improbcat, 2025-07-23:
I fat32 formatted the card and put your mpeg directory on it. 
Didn't know there even was a serial monitor, re-uploaded and serial monitor shows nothing but this:
--
JP�R
--

thelastoutpostworkshop, 2025-07-23:
Make sure the Serial monitor is set at 115200 baud :

[image]

Improbcat, 2025-07-23:
Changed to 115200 baud. Pressing the reset button on the card gets this in serial monitor:
--
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4888
load:0x40078000,len:16516
load:0x40080400,len:4
load:0x40080404,len:3476
entry 0x400805b4
Display initialization
Screeen size Width=240,Height=320
SD Card initialization
ERROR: File system mount failed!
--

When compiling and uploading nothing shows up in the serial monitor. It stays blank.

Improbcat, 2025-07-23:
[image]
[image]

[This is the board I bought](https://www.aliexpress.us/item/3256808025838125.html)
[This is the manufacturer's website linked from aliexpress](https://www.tztstore.com/goods/show-7983.html) Under download is a zip with information including what looks like they might be pinout diagrams (again I am brand new at this, so I am slightly guessing).

Thank you for all your help with this. I really appreciate you taking your time to help. I thought this would be an easy project to get started with CYD and this sort of electronics in general, and did not realize there was such a variety of boards sold under the same or similar model numbers.

thelastoutpostworkshop, 2025-07-23:
> (thelastoutpostworkshop/esp32-2432S028_video_player#6)
>
> I just discovered that the SD card formatter formatted the card in exFAT
> and not FAT32. It doesn't seem to have an option to format in FAT32
> (possibly because I am using a 64gb card?). Just more data that might be
> relevant.

yes good catch, the sd card reader works only with fat32, try a 32gb card
or less

Improbcat, 2025-07-23:
PROGRESS!! 
I formated a 32 card as FAT32, put the mpeg folder on it, and tried again. Serial monitor is showing it as apparently trying to play the videos, screen is not lighting up. So we seem to be down to a video problem.
The screen is labeled L1435 2.4

thelastoutpostworkshop, 2025-07-23:
Very good, let me check the pinout of this board

thelastoutpostworkshop, 2025-07-23:
The only line that need to be changed is the BL pin.  
Change this line : 
`#define BL_PIN 21`
to this:
`#define BL_PIN 27
`