thelastoutpostworkshop/esp32-2432S028_video_player #7: bright colors

StaybanD, 2025-07-24:
I have something wrong with bright colors. Help please.
all colors are suitable, but some have artifacts, I don't understand what's wrong

[image]

thelastoutpostworkshop, 2025-07-25:
Yes i have the exact same thing, its due to the compression, every frame is compressed in jpeg and stored in mjpeg.

thelastoutpostworkshop, 2025-07-30:
> (thelastoutpostworkshop/esp32-2432S028_video_player#7)
>
> I used 4 different programs to convert videos, I also used different
> settings, but the result did not change

it’s a combination of image compression and color capabilities of the
display controller.  I do not think it can be improved much based on your
experimentd

thelastoutpostworkshop, 2025-08-27:
Thanks for investigating the issue.

gbouthenot, 2025-08-28:
I tried to use [TJpg_Decoder](https://github.com/Bodmer/TJpg_Decoder/) instead of [ JPEGDecoder (same author)](https://github.com/Bodmer/JPEGDecoder), because the author said:
>News: October 2019 - I have created a new Arduino compatible [Jpeg decoder library here.](https://github.com/Bodmer/TJpg_Decoder) This new library is based on TinyJpegDec, this is 60% faster on 32bit processors, it also has a much simpler interface.

But it has the same flaws and is even slower !

With JpegDec:
```
Playing /mjpeg/VID-20250827-WA0001-q8.mjpeg
Opening /mjpeg/VID-20250827-WA0001-q8.mjpeg
MJPEG start
MJPEG end
Total frames: 97
Time used: 6046 ms
Average FPS: 16.0
Read MJPEG: 730 ms (12.1 %)
Decode video: 3355 ms (55.5 %)
Show video: 1961 ms (32.4 %)
```

```
With TJpg_Decoder:
Playing /mjpeg/VID-20250827-WA0000-q8.mjpeg
Opening /mjpeg/VID-20250827-WA0000-q8.mjpeg
MJPEG start
MJPEG end
Total frames: 97
Time used: 8681 ms
Average FPS: 11.2
Read MJPEG: 803 ms (9.3 %)
Decode video: 5360 ms (61.7 %)
Show video: 2518 ms (29.0 %)
```

(I don't know why `Show video` is also slower (and the screen shows a lot more glitches)