thelastoutpostworkshop/esp32-2432S028_video_player #8: Adding sound?

DynaMight1124, 2025-07-26:
Hi

Great project, works really well. I know the CYD sound isnt great however with sound mods its actually not all that bad and also apparently the newer CYD2USB revisions have the sound fixed from the factory.

Mod: https://github.com/hexeguitar/ESP32_TFT_PIO?tab=readme-ov-file#audio-amp-gain-mod 

Would it be possible to add sound support into the project? 

Thanks

thelastoutpostworkshop, 2025-07-26:
Thanks for sharing the mod.
By adding sound, do you mean the sound included in a video ?

DynaMight1124, 2025-07-26:
Hi, I could be wrong but I assumed that even if a video was encoded with sound the CYD would not play sound? I couldnt see any sound related code in the sketch. Apologies if I'm wrong.

thelastoutpostworkshop, 2025-07-26:
Yes, but playing sound ability could be added

ubear7, 2025-08-12:
Could you please show explain how. Thanks.

closed 2025-11-29 by thelastoutpostworkshop