thelastoutpostworkshop/esp32-2432S028_video_player #9: Problème d'affichage

Lucien524, 2025-07-29:
Bonjour,

[image]

Problème d'affichage. Les couleurrs sont bonnes.
Que peut on faire ?
Merci !

Lucien524, 2025-07-30:
Oui, j'utilise bien (Désinstallés puis Réinstallés) :
IDE 2.3.6
ESP32 Dev Module 3.2.0
GFX library.h 1.6.0
JPEGDEC 1.8.2
 Résultat identique à la photo...

Lucien524, 2025-07-30:
[image]

La video est bien une de celles fournies avec le projet
La video est correcte au visionnage

thelastoutpostworkshop, 2025-07-30:
Que voyez-vous sur le Serial Monitor dans l'IDE Arduino ?

thelastoutpostworkshop, 2025-07-30:
> (thelastoutpostworkshop/esp32-2432S028_video_player#9)
>
> IMG_9626.JPG (view on web)

je vois que les mjpeg ont des erreurs de lecture, essayer de reformatter la
carte SD et recopier les fichiers

Lucien524, 2025-07-30:
Formatage refait pour le Xième fois, pas d'amélioration...

thelastoutpostworkshop, 2025-08-13:
For a another user, this modification works:
`#define DISPLAY_SPI_SPEED 40000000L // 40MHz `