thelastoutpostworkshop/esp32-2432S028_video_player #13: Disordered picture

akaditya13, 2025-08-12:
hi, i have countered an issue. my display is ESP32-2432S028. after uploading, the picture is displayed but somehow disordered

https://github.com/user-attachments/assets/48ab3285-ca1a-4932-b65b-ec4ee3092cf1

. what may cause the issue?

thelastoutpostworkshop, 2025-08-13:
Great! let me take a look

thelastoutpostworkshop, 2025-08-13:
pl let's try replacing the same line, with this :
`Arduino_DataBus *bus = new Arduino_ESP32SPI(2 /* DC */, 15 /* CS */, 14 /* SCK */, 13 /* MOSI */, 12 /* MISO */);
`

akaditya13, 2025-08-13:
somehow still scrambled

akaditya13, 2025-08-13:
i got it from here
[](url)https://randomnerdtutorials.com/lvgl-cheap-yellow-display-esp32-2432s028r/
he said that with an ILI9341 display 40MHz works OK, 80MHz sometimes fails.
i guess it only need this line:
`#define DISPLAY_SPI_SPEED 40000000L // 40MHz`

thelastoutpostworkshop, 2025-08-13:
Very good thanks for the investigation.
So does it work also with this original bus configuration ?
`Arduino_DataBus *bus = new Arduino_HWSPI(2 /* DC */, 15 /* CS */, 14 /* SCK */, 13 /* MOSI */, 12 /* MISO */); 
`

akaditya13, 2025-08-13:
I tried one by one, all three works fine

closed 2025-08-13 by thelastoutpostworkshop