thelastoutpostworkshop/esp32-2432S028_video_player #20: model 2432S022 = White Screen

theawesome9000, 2025-09-06:
Good evening,

everything you posted loaded cleanly, and serial shows it is cycling between videos but the screen is blank white. I assume it's a different pinout or needs a different library because of the touchscreen. any help would be great!

thelastoutpostworkshop, 2025-09-06:
Yes it's a different model, do you have link to your model ?

theawesome9000, 2025-09-06:
Yes sir, https://a.co/d/bj0ROqBSent from my iPhone

thelastoutpostworkshop, 2025-09-06:
> <thelastoutpostworkshop/esp32-2432S028_video_player/issues/20/3262149748@
> github.com>

the link leads to nothing

theawesome9000, 2025-09-06:
I'm so sorry, try this one.

[[https://a.co/d/dFS1itc](url)](https://www.amazon.com/DIYmalls-ESP32-2432S022C-Touchscreen-ESP32-WROOM-32-Development/dp/B0DH1P13DW?crid=1K02OABOH5HV6&dib=eyJ2IjoiMSJ9.p99zwUSQP2ELEv7-Q0epYxlNwYd5YEXlGSbYOagRfhac-c1IlhBcBzOj2-IQtVlAb5lyvDXrTHf-qs6rm60T4fJgYbFtHHEs7pG4FKTPhneDWO4YdWQ4jtqz5NGSh2NM_fru6lmKopAxc_PGcYsCrt4FVHvn--Dd3jtuFLZUwwapQIQCWT1Y46sWKCM8w2kwIgZralUOaR485mGBdPObjZfh6S0eTFhs-ilwJShCf50.JV96Qzk5l7x0QkOJc1RzjkKBa2ALq9OwjNzOK1NB0nQ&dib_tag=se&keywords=2.2+inch+esp32&qid=1757165311&sprefix=2.2+inch+esp32%2Caps%2C88&sr=8-3)

thelastoutpostworkshop, 2025-09-06:
ok this work, let me take a look

thelastoutpostworkshop, 2025-09-06:
It will not work with this project, it's a different display controller (ST7789)

theawesome9000, 2025-09-06:
That explains it ☹️ 
Thanks for checking this out

closed 2025-09-06 by thelastoutpostworkshop